satnogs/satnogs-network #102: Test client connectivity

ppapadeas, 2015-03-31:
We need to implement a "test client connectivity" for each ground station that is registered in our network. This will be useful for two User stories:

- Verification of a new ground station. A user adds a ground station and then when ready clicks on "test client connectivity". The test is run and then automatically the ground station is marked as active.

- User or admin invoked test (even periodically for GS not utilized for long time).

On the test itself: Once the test in invoked a job (orphaned or tied to a test observation) is created for a test satellite for 1 minute "pass" within the next 30 minutes. Job is scheduled, client picks it up, runs it, records it and sends back the payload. Once the payload is uploaded for a test job on network side then Ground Station is set as active.

On setting a ground station as active: Admins can toggle activity from /admin. Users can set a ground station as inactive with a click. For users to set a GS as active they need to run the "test client connectivity" and that sets the GS as active automatically.

On historical runs: Tests (with results = payload or not) should be displayed on a separate tab on ground station page.

jinxcat, 2015-03-31:
Very much needed functionality.

Shouldn't there also be a mechanism to uniquely identify satnogs devices?

johngian, 2015-03-31:
It's probably better this to be implemented on the client side. Network-to-client communications are much more difficult (dynamic IPs, NAT). It makes sense to me to add a periodic-health test in the ground station and gather logs/metrics somehow (ELK probably).

jinxcat, 2015-03-31:
I would say add both

ppapadeas, 2015-03-31:
@johngian Read the specs again. It will not be netowrk-to-client. +1 on periodic health on client, but that should be tracked on the client repo.

johngian, 2015-03-31:
Hm, yes that makes sense.
Let's keep the client health/logging logic in the other repo.

+1 for that functionality

jinxcat, 2015-03-31:
This functionality (of this issue) should be ran periodically as well I believe

ppapadeas, 2015-03-31:
@jinxcat check user story # 2 already there.

comzeradd, 2015-04-08:
We decided on a different much simpler approach at yesterday's meeting. @ppapadeas @johngian could you elaborate on the specifics?

comzeradd, 2015-04-20:
Ping @johngian @ppapadeas 

Can you elaborate on specs? Do we need to require the Jobs endpoint to only work with API key?

commit d1c2773b (2015-04-07): [Fixes #102] Test client connectivity
commit f9613a39 (2015-04-09): [Fixes #102] Test client connectivity
commit 66963863 (2015-04-09): [Fixes #102] Test client connectivity
commit 68330493 (2015-04-19): [Fixes #102] Test client connectivity
closed 2015-05-06 by comzeradd