satnogs/satnogs-network #106: Determine how to handle Ground Station operators

comzeradd, 2015-04-04:
Currently whoever adds a Ground Station is listed as owner. We need a smarter way to handle cases where a Ground Station is operated by an Organization, University, etc.

Suggestion:
1. Change `owner` to `ManyToMany`, so more users can handle the same Ground Station.
2. Add `operator` free text field, visible to Station view.

jinxcat, 2015-04-04:
Why free text?

comzeradd, 2015-04-04:
What else? People create individual accounts to add organizational Ground Stations.

jinxcat, 2015-04-04:
Doesn't sound right.... We should have a better data structure around that I think.

Free text is unusable

comzeradd, 2015-04-04:
Yes, I agree. We can't link to a profile or something similar.

ppapadeas, 2015-05-03:
Org accounts would be nice to be handled like old GitHub ones. (basically uplevel the account). Couldn't it be a toggle on the profile? That would then expose additional information? At the end of the day *someone* owes to be responsible for the Ground Station (even when operating in clubs/unis/etc)

comzeradd, 2015-07-22:
One solution to this issue would be to change the relation between users and ground stations, so that on station can have multiple administrators.

jinxcat, 2015-07-22:
That's not a bad idea. And it should not add more functionality (adding,
deleting) other than the operators being a list instead of a monad

ryanturner, 2016-02-23:
This is a feature that I'm particularly willing to play guinea pig for. My organization [HamWAN Memphis Metro](http://www.memhamwan.org) is building a station, and we're particularly interested in allowing our membership to utilize the system. Since we'd be potentially giving access to new, inexperienced users, it would be beneficial to be able to limit access of these child/sibling accounts to specific nodes. I'd hate for one of our new users to cause issues for existing folks. In addition, I wouldn't want these users to be able to fully administrate the station -- there are definite users who are network/systems administrators, and there are other users who are just members.

faulteh, 2016-12-21:
With the addition of a new ground station you could add a new permissions eg 'can_manage_groundstation_PK' (perhaps via a management call_function as I've seen done similarly before), then additional users can be added to a group/teams with these permissions.

Perhaps adding django guardian you get per-object permissions to groups, so that modification of observation/data records gets naturally limited to these groups without endangering observation/data objects in the same model but owned by others.