satnogs/satnogs-network #128: Support for fixed stations

cshields, 2015-05-27:
Some people may want to participate in SatNOGS but only have means for a fixed station.  Such a fixed station would be very limited in comparison to a full AZ/EL rotor station, but is better than no station at all.

This was the idea behind entry into carpcomm, a simple yagi pointed 30 degrees up and +/- 30 degrees AZ to catch passes as they ascend/descend:

[image]
(carpcomm antenna from http://blog.carpcomm.com/)

To support this in the network would just be some configuration that notes it is a fixed station, and what direction it is facing for observation calculations   (this could also tie in with the last 2 issues I just filed, #126 #127 ).  Those calculations would just have to account for a much smaller, directional window which may or may not be easy to do.

If we decide to go with this, satnogs-client would need a configuration to skip all rotor control as well, otherwise everything else would be the same.

cshields, 2015-05-27:
I should also add that this does not have to be a fixed-yagi either..  Some people have eggbeater antennas and others that are omnidirectional and could serve SatNOGS with data, without needing a rotor.  In this case, there would be no need for a change in satnogs-network, just a need to ignore rotor controls in satnogs-client.

papamat, 2015-05-27:
Is good to consider azimuth only rotators too.

cshields, 2016-04-28:
fixed in v0.2.5

closed 2016-04-28 by cshields