satnogs/satnogs-network #140: Implement the ability to comment/annotate observations

cshields, 2015-07-21:
It would be excellent if vetted accounts (to prevent spam) could comment on observations.  For instance, after a successful capture today I wanted to be able to annotate with conditions (ie: the fact that there was no LNA involved this time).

Thinking forward when an observation may involve a dozen or more stations, a discussion should be had around whether these comments would attach to the observation as a whole or the individual station data captures.

And, to complete this feature, a small icon indicating a comment thread should be added in the Observations listing.

Thoughts?

comzeradd, 2015-07-22:
Having some short of discussion per observation is indeed a nice idea. Not just for annotating stuff, but also as a general discussion opportunity initiated by observers and not just by operators (eg. why I scheduled this observation, how to decode it, etc).

But instead of developing a complete discussion functionality inside the network app we could probably utilize discourse (the [community forum](https://community.satnogs.org/)). This is shaped by two things:
1. Link to "New Topic" with a predefined subject in a predefined category. This seems to be [supported](https://meta.discourse.org/t/compose-a-new-pre-filled-topic-via-url/28074?u=techapj) by discourse.
2. Ask the forum API if the topic already exists and link to it instead of pointing to the create form.

jinxcat, 2015-07-22:
Using discourse might be a somewhat practical solution, but it is
absolutely external to the satnogs ecosystem imho.

If we decide to start keeping metadata on (solitary or concatenated)
observations, it should be in open and simple format; probably something
like JSON. discourse offers no such thing.

Something like soundcloud comments comes to mind as quite interesting

comzeradd, 2015-07-22:
> If we decide to start keeping metadata on (solitary or concatenated)
> observations, it should be in open and simple format; probably something
> like JSON. discourse offers no such thing.

It [does](https://community.satnogs.org/t/345.json). So if we ever change our mind we could migrate the discussions.

The soundcloud example seems nice, but I'm not sure we want such detailed comments at the moment.

comzeradd, 2015-08-28:
So here is an attempted implementation. We add a "Discussion" button to each Observation view and we detect if there is already a topic. If it's already there we just link to it (in the future we may display the number of comments it has). If it doesn't exist we link to the "New Topic" dialog of the forum with prepopulated Title, Category and some part of the body.

See the screencast below for preview. On the first run we see how a new topic is created. On the second one we just link to it.

[image]

comzeradd, 2016-01-17:
Deployed on production

commit 66779b02 (2015-08-28): [Fixes #140] Add Discussion button on Observations
closed 2016-01-17 by comzeradd