satnogs/satnogs-network #171: indication for "received data" or "no received data"

cshields, 2015-10-17:
We have the red/green indication for successful observation data collection.  Beyond that just looking through the feed of observations there is currently no way to tell which of those successful uploads are good and which have nothing received.  To highlight the problem, I have had probably half a dozen successful captures in the last couple of days and about just as many duds.  Anyone who visits the network looking for data has to listen to each one to know whether or not something is good.

Today this could be a human process, someone listens to the stream and verifies that there is something in it (cw, packets, etc) or verifies that there is nothing of interest.  Long down the road if we get into automatic decoding the network could also verify.

I would recommend a simple addition to the colored ID number with a "verified" icon to the feed, or a :+1: for known good and :-1: for known bad.   From the observation view itself this would require addition of such an input for logged in users.

I would also recommend going a step further and setting a decay lifespan on observations that are flagged as duds (but not on observations that have no flag set)... but that can be a different issue.

ppapadeas, 2015-10-17:
That is a great suggestion @cshields ! @comzeradd what do you think?
In terms of permissions I would say the observer, the GS owner (of the data) and any admins would be OK.

Visually we can amend the current coloring scheme. Green would be verified, light green not? (or something like that)

For the last part of your comment @cshields I would suggest that being a separate issue/discussion indeed.

comzeradd, 2016-03-24:
So I see four levels of status:
* blue: future (pending)
* red: has no data at all
* orange: has data but not verified
* green: has verified data

If we restrict this to the three roles @ppapadeas mentions above (observer, GS owner, admins) I'd say one upvote is enough for an observation to be considered verified.

So next to each Data block title could have the `glyphicon-thumbs-up` icon. And we can use the `glyphicon-ok-sign` for already verified data.

Just to keep it simple for now I would suggest we only have an upvote functionality, since observation data that haven't received any "thumbs up" would remain in an unverified stage and we currently don't think of deleting data. No need to downvote for now.

cshields, 2016-03-24:
thought on signal quality?  ie: marginal vs good? A good signal, in the future, may automatically be decoded while a marginal signal may exist but need manual intervention (like decoding CW by ear).

I guess in this scenario through automation we would go straight to green on a good signal, and marginal signals would remain blue/unknown until someone comes along to mark them as having data but not verified.

cshields, 2016-03-24:
also, I suggest s/blue/gray/ as that is what we have across the board today and looks good. We can shade the rows / headers a light red/green/orange for the other settings.

Agreed on the first pass being "verify only", though I will add the other options to the model in a CharField.

comzeradd, 2016-03-24:
I agree about the color scheme. Gray would probably look better.

I haven't given much thought about signal quality, but if we have a way in the future to determine its quality and immediately mark data as verified it would save us a lot of time.

cshields, 2016-03-25:
Some initial work on this tonight I hope to have committed in the next 24hr (pending some feedback here). I went ahead and added the ability to mark 'no data' so people do not waste time sifting through data that is 'not known good'. I feel this is critical as a lot of it is no data today, especially in -dev.

@comzeradd @ppapadeas 

[image]

comzeradd, 2016-03-25:
Nice :) The colors may be too much. I feel it's too much distraction away from the actual data. We can use the color scheme in the observation list to color observation IDs.

I would just leave the text and a verified checkmark. For instance something like that (using glyphicaons instead):
:heavy_check_mark: verified
:heavy_multiplication_x: no valid data
and either add the metadata (who, when) next to it, or just on a title element as a tooltip.

cshields, 2016-03-25:
I still feel that some kind of color helps in visually scanning. How about this?

Also - in the observation list view we have red/green colored ID numbers if there is data uploaded which does not take into account if any of that data is verified. We will want a new scheme for that as well so the observation list can be visually filtered (a filter option itself would be good down the road too).

[image]

cshields, 2016-03-25:
For the observation list,

Current state:
Future: light blue
No uploaded data: red
Uploaded data: green

My proposed scheme
Future: light blue (same)
No uploaded data and/or all vetted bad data: red
Some or all unvetted data with no verified good data: orange
Some or all verified good data: green

comzeradd, 2016-03-25:
Yes, in my initial comment with color suggestions I was actually referring to the IDs background on the observation list ;)

The new screenshot looks really nice :+1: You could try the [glyphicons](http://getbootstrap.com/components/#glyphicons) with the filled background. `glyphicon-ok-sign` instead of `glyphicon-ok`, to check if these look better. But I think it generally looks good.

cshields, 2016-03-25:
I'm mixed on the -sign icons. Will leave it up to you:

[image]

etihwnad, 2016-03-25:
Minor style niggles from the peanut gallery.

The bars above and below the waveform blend into the adjacent bars from the
next station.

Maybe a single 2-line? bar either above below the waveform, with a frame
around the station's data.

Transport/play controls are usually on the left.  C.f. video players etc.

The X/check can be fully left, this may make it easier to scan the list for
status.  After the check is the station, then time range.  Or a different
order that is the natural progression of info someone would be interested
in when drilling down into the data.

cshields, 2016-03-25:
We could probably improve the blending by increasing the border size a tad.

Here is what the icons look like after a pull-left:

[image]

comzeradd, 2016-03-25:
I suggest we focus on the backend functionality. UI is a good start and we could revisit once we have the whole functionality working.

cshields, 2016-03-25:
> I suggest we focus on the backend functionality. UI is a good start and we could revisit once we have the whole functionality working.

Backend is all done.  :)  I added changes to the list as well, and added visibility restrictions this morning (is_staff, observation owner, station owner per pierros).

Will clean up a bit and commit later.

comzeradd, 2016-04-23:
Deployed on production

closed 2016-04-23 by comzeradd