satnogs/satnogs-network #186: Ground Station Edit/Add Field Validation

angrymeerkat, 2016-01-02:
https://community.satnogs.org/t/problem-adding-ground-station/521

Field validation should be displayed or shown before allowing a ground station to be added. This stems from the altitude field accepting decimal input, but causing an add/edit failure due to the data type being an integer. This causes the request message to return an error.

angrymeerkat, 2016-01-03:
[image]

Potential stopgap measure. I'm not sure how to add input validation during data entry, but this seem to be more descriptive than the "Some fields .." error message.

cshields, 2016-02-06:
In addition, if someone enters anything in addition to the integer for altitude (ie: "100m"), the error returned is that some fields are missing.  Should catch that this is something other than the integer and call it out.  See also https://community.satnogs.org/t/add-station-problem/562

[image]

ppapadeas, 2016-02-07:
@comzeradd can you have a look at this?

ryanturner, 2016-03-08:
This impacted me as well and resulted in a significant delay when trying to create my station.

comzeradd, 2016-04-23:
I opened a PR for this. I added relevant attributes on the form (required and type) where applicable. So any modern browser (any but Safari) won't let you submit the form with errors. Additionally form errors are displayed on the alert message to help you correct them.

closed 2016-10-11 by ppapadeas