satnogs/satnogs-network #190: Mhz vs Hz on Ground station list

ppapadeas, 2016-01-18:
https://network.satnogs.org/stations/

On antenna listing the Mhz should be displayed correctly.

robberwick, 2016-02-21:
I was thinking of taking a look at this, but I'm unsure as to what the exact issue is. Currently, frequency values are displayed with a "MHz" suffix throughout the app. Is this not the desired format? Is the requirement that all antenna frequencies displayed with the suffix "Mhz"?

ppapadeas, 2016-02-21:
That was more of a reminder issue, to make sure that whenever we display frequencies we store them internally as hz and then format them in MHz on our UI.

@robberwick Can you make a swipe, making sure that this is applied universally on our app? Thanks!

robberwick, 2016-02-21:
@ppapadeas I've had a look, and certainly the development fixtures install 3x `Antenna` objects where the frequencies are stated in MHz (e.g. 144.8), so the `frq` custom template causes them to be rendered as "000.000" because it divides the value by 1000000 and formats to 3 decimal places. However, that behaviour is not present when I look at the live site - frequency figures appear to be displayed correctly, which suggests that the values are already stored in the db in Hz, thus surviving the `frq` template filter.

So, assuming that is correct (is it possible to check?), it seems that the job is to:
* Update `fixtures.json` to populate the `Antenna` objects with values in Hz
* Migrate the `frequency` field to a `PositiveIntegerField` so that users are not tempted to enter a float
* Optionally add a `help_text` property to the field explaining that values should be entered in Hz

If that assumption is incorrect, and the frequency values in the db *are* stored in Mhz, then we would also need to apply a data migration to massage the values appropriately. Again, it would be useful to know what the current state of the data is.

comzeradd, 2016-04-23:
I opened a [PR](https://github.com/satnogs/satnogs-network/pull/244) to fix the fixtures.

On production we didn't use the fixtures that's why we see these correctly. Also we don't expose anywhere on the UI the ability for users to add Antennas themselves, so I don't think we need any other fix for now. If this is the only occurrence of wrong frequencies we are good with this PR.

closed 2016-10-11 by ppapadeas