satnogs/satnogs-network #228: unavailable stations still show up in an observation

cshields, 2016-03-27:
When we schedule an observation, stations that are passed but unavailable will show in the timeline with no bar showing their availability. This is fine. However, when scheduling such an observation those unavailable stations will still show up in the scheduled observation view. See image below.

This should be a rare case, I triggered it by scheduling [an observation](https://network-dev.satnogs.org/observations/360/) that I intended for my omni station but it was not marked operational. After I realized this, I marked my omni as operational and went to submit [a new one](https://network-dev.satnogs.org/observations/361/) - expecting that the previous scheduled stations would be excluded. Maybe not so rare as we get more stations and passes that overlap.

In the screenshot below, bot KB9JHU and MemHamWAN are previously scheduled and unavailable for this observation.

[image]

comzeradd, 2016-04-29:
If I got this right, on this specific screenshot, besides the omni station, the other two have observation windows that are excluded since they overlap with previous scheduled observation, right?

cshields, 2016-04-29:
correct