satnogs/satnogs-network #249: stale ground stations appear in observations

cshields, 2016-04-28:
There are a couple of ground stations in dev that are not providing any data, one was last seen 4 days ago and the other 6 days ago. We should exclude these if they have not been seen in... 24 hours?

See also https://github.com/satnogs/satnogs-network/issues/228

comzeradd, 2016-04-28:
We have the `STATION_HEARTBEAT_TIME` in [settings](https://github.com/satnogs/satnogs-network/blob/dev/network/settings/base.py#L212).

On production is 60 minutes, on dev is 10 days. Maybe 10 days is too much?

cshields, 2016-04-28:
Yeah, I'd suggest 1 day..  60 mins is a good idea for prod. Closing this out.

closed 2016-04-28 by cshields

comzeradd, 2016-04-28:
I pushed a [fix](https://github.com/satnogs/satnogs-ansible/commit/577f83016d5ea36c5a253df0a65b03c303439c03). It would roll out on the next deployment.