satnogs/satnogs-network #255: Show satellite pass attirbutes on observations

comzeradd, 2016-05-06:
While calculating next passes for a given satellite the [library](http://rhodesmill.org/pyephem/quick.html) we use (pyephem) offers all related info:
```
0  Rise time
1  Rise azimuth
2  Maximum altitude time
3  Maximum altitude
4  Set time
5  Set azimuth
```
It would be nice to display these attributes either during scheduling a new observation or on already scheduled ones.

cshields, 2017-01-27:
loosely related: satnogs-network#278

darksidelemm, 2017-07-30:
It would be useful to see the pass plot (what is shown on the scheduling page), as this gives more information than just start/stop azimuth and peak elevation. It would certainly help me diagnose some antenna shadowing issues.

cshields, 2017-07-30:
<darkside> the real wanky approach woudl be to have a vertical graph showing elevation & azimuth alongside the waterfall, but that might be going a bit too far :-)

I like that idea, albeit more involved..

kerel-fs, 2017-07-30:
This would also be possible with a client-side solution, using [gpredict-js](https://github.com/kerel-fs/gpredict.js) or [satellite-js](https://github.com/shashwatak/satellite-js) and TLE given by satnogs-network.
The latter has a wider user base, while gpredict-js is imho easier to use, but has only two users ([phonesat](https://github.com/cepr/phonesat-website), who ported it to js, and soon satnogs-db via satnogs/satnogs-db#159).
In satnogs-db, the orbit will be calculated from TLE on the client-side, reducing network usage for status updates and enabling more frequent position updates as well as displaying a ground track.