satnogs/satnogs-network #276: Support for waterfall images from clients

cshields, 2017-01-15:
work is underway in the client to generate waterfall images as a capture is made. We need to be able to support the submission of these in /api/data/ and the display in an observation window.

comzeradd, 2017-01-21:
Besides the UI work, does this require something more than just an extra FileField on the Data model?

cshields, 2017-01-21:
That's how I would envision it, yes.

I'm not sure how the waterfall work is going, that's in gr-satnogs#42 so its hard to say size and shape for now but I would imagine a very detailed image, which we could display in a reasonable size with some mouseover zooming capability?

So for now we can expect a file would be uploaded and linked in the DB, so another FileField and support in the API would be a start.

@surligas

ppapadeas, 2017-01-23:
We just discussed it on the weekly meeting. @comzeradd please proceed with implementation

cshields, 2017-01-27:
working very well on dev!  All of the waterfall images from overnight observations (before API support) were automatically uploaded through the client to dev as it went through the backlog.

comzeradd, 2017-01-27:
That's awesome! I opened #288 to rethink the UI.

closed 2017-01-27 by comzeradd