satnogs/satnogs-network #314: Empty waveform and spectrogram when they are initialized before user visits audio tab

adamkalis, 2017-02-14:
STR:

1. Open an observation
2. Wait until ogg file is loaded
3. Wait some more seconds for browser to analyzing the data and try to create waveform and spectrogram
4. Visit audio tab

Expected:
User should see waveform and spectrogram

Actual result:
User sees empty waveform and spectrogram
======================
I have noticed that in case that waveform and spectrogram containers are not displayed (display: none on them or their parent), they can not (not sure why) load data into their canvas.

A quick hack could be to use "opacity: 0" instead "display: none".

cshields, 2017-05-21:
we should prioritize this, new users have run into this bug this weekend..

comzeradd, 2017-05-21:
It seems to be fixed on dev after 3b184701252fdacc4a3258cfe396edaf664e5ac9

@adamkalis can you confirm?

adamkalis, 2017-05-22:
@comzeradd Yes I confirm that this is fixed in dev.

closed 2017-05-22 by comzeradd