satnogs/satnogs-network #315: Observations page takes too long to load

ppapadeas, 2017-02-21:
[image]

~8sec to get a response from the server. We should re-consider how we serve the page.

faulteh, 2017-02-21:
- Currently filtering by property which can't be done by SQL so the python is slowing it down
- Change the query to prefetch the datas at the time of the main query
- Since the resultset is not likely to change extremely often but the result set will grow over time might look at caching the result for 5-10 minutes and/or until it's stale (cache max obs id > db max obs id) for the default case of 'no filtering'

If you assign it to me I'll take a look at it this evening and tomorrow.

ppapadeas, 2017-02-21:
@faulteh good thoughts. Feel free to take this over and see where we can get. On caching I would be cautious as we may want to have kinda live results on the observations page. A smaller interval might be better. (or even celery it out?)

commit 0d77fd7e (2017-02-24): [Fixes #315] Use query annotation instead of model properties to filter Observation faster.
commit 573a0f43 (2017-02-24): Merge pull request #317 from faulteh/obsview-perf

[Fixes #315] Use query annotation instead of model properties to filter Observation faster
closed 2017-02-24 by comzeradd