satnogs/satnogs-network #387: Can't Calculate Observations for 40967-Fox1A because there are no TLEs for this satellite yet

ka2upw, 2017-08-03:
Can't Calculate Observations for 40967-Fox1A.  When you click on the "Calculate Observation" button the system tells you "No TLEs for this satellite yet." and you can't schedule any observations.  The TLEs are available from AMSAT at 

https://www.amsat.org/amsat/ftp/keps/current/nasa.all 

but this particular satellite's TLEs are probably not available from other sources due to the nature of the launch.

comzeradd, 2017-08-04:
I manually added TLE from AMSAT and I was able to schedule an [observation](https://network.satnogs.org/observations/11654/).

Thanks for reporting! 🍨

closed 2017-08-04 by comzeradd

ppapadeas, 2017-08-04:
We need a more holistic approach to it: https://github.com/satnogs/satnogs-network/issues/388

ka2upw, 2017-08-04:
Hi Nikos,

I tried to schedule an observation of 40967 Fox-1a on network-dev and it still says "No TLEs for this satellite yet."

Were the TLEs added only on "network" but not on "network-dev"?

Regards,
Douglas

comzeradd, 2017-08-05:
> Were the TLEs added only on "network" but not on "network-dev"?

Yes, I only added the TLEs on the production instance.